CreatorDev/contiki #8: Add an I2C release function to the PIC32 I2C driver

mtusnio, 2016-11-28:
Right now there's no way to deinit or release the I2C interface after initing, some function to do that would be useful.

francois-berder-imgtec, 2016-12-15:
No need for that function because it would be very similar to i2c_init except that it would not set SMEN bit.

Shpinkso, 2017-01-03:
@mtusnio is this still an open issue in your opinion?

mtusnio, 2017-01-03:
Yes, it's not a high priority, the function should be simple enough (one bit switch), but it'd make sense to keep a release even if we just want to make it future proof in case something more does need to go into there.

francois-berder-imgtec, 2017-02-09:
The i2c_release function needs to:
*  disable the i2c module
* unregister i2c from lpm module
* power down i2c

Blocked until CreatorDev/contiki#29 is done.

closed 2017-03-21 by mtusnio