CreatorDev/contiki #14: Investigate cascoda driver issues with contiki master

abhijit-mahajani-imgtec, 2017-01-09:
AS a result of #11, its observed that there are some issues with cascoda driver working with contiki-master (upstream version)

mtusnio, 2017-01-10:
https://github.com/mtusnio/contiki/tree/upstream_merge - upstream merged into 3.0-mikro-e with a fix for the clock, ca8210-conf.h will need the following change:

diff --git a/platform/mikro-e/ca8210-conf.h b/platform/mikro-e/ca8210-conf.h
index 9283e83..6ec21f8 100644
--- a/platform/mikro-e/ca8210-conf.h
+++ b/platform/mikro-e/ca8210-conf.h
@@ -36,9 +36,9 @@
 #ifndef CA8210_CONF_H
 #define CA8210_CONF_H
 
-#undef DEBUG /* Conflict with a macro named DEBUG defined in compiler */
-#include <p32xxxx.h>
 #undef DEBUG
+#include <p32xxxx.h>
+#define DEBUG 0
 
 #include <pic32_gpio.h>
 #include <pic32_irq.h>

francois-berder-imgtec, 2017-01-13:
RPL is now enabled by default. RPL is currently not working no udp connection from clicker to ci40. Disabling it enables NA which re-establish the previous behaviour before the merge with upstream.