CreatorDev/contiki #15: Fix undef of DEBUG macro while building with upstream merged in

mtusnio, 2017-01-09:
Our ca8210-conf.h & cc2520-conf.h files do undef DEBUG while including the p32xxxx.h file, however upstream is using the macro in some if conditions. We either need to simply define the macro again with some value, or find a way to either avoid having to undefine it or to save and restore the value.

francois-berder-imgtec, 2017-01-13:
One possible solution would be:
```
diff --git a/core/cfs/cfs-coffee.c b/core/cfs/cfs-coffee.c
index 9aa5726..79d7b41 100644
--- a/core/cfs/cfs-coffee.c
+++ b/core/cfs/cfs-coffee.c
@@ -46,6 +46,12 @@
 #include <limits.h>
 #include <string.h>
 
+
+#include "contiki-conf.h"
+#include "cfs/cfs.h"
+#include "cfs-coffee-arch.h"
+#include "cfs/cfs-coffee.h"
+
 #define DEBUG 0
 #if DEBUG
 #include <stdio.h>
@@ -54,10 +60,6 @@
 #define PRINTF(...)
 #endif
 
-#include "contiki-conf.h"
-#include "cfs/cfs.h"
-#include "cfs-coffee-arch.h"
-#include "cfs/cfs-coffee.h"
 
 /* Micro logs enable modifications on storage types that do not support
    in-place updates. This applies primarily to flash memories. */
```

francois-berder-imgtec, 2017-02-09:
The pull request attached to this issue only works for cascoda clickers.

Shpinkso, 2017-02-10:
Is this blocked on #49 ?

closed 2017-02-14 by francois-berder-imgtec