CreatorDev/contiki #46: Automated update mechanism

Shpinkso, 2017-02-09:
We want to be able to auto-update a clicker with the latest build output on CI
Acceptance criteria:
- Evaluate most reliable s/w update mechanism triggered from a Jenkins client (Linux machine)
Updating via an intermediate device (e.g. Rpi) is acceptable, if reliable.
- Mechanism should require minimal (ideally no) despoke build steps/configurations for Contiki
- Provide some stats of % update failure over a soak test
- Confirm hardware requirements to purchase (if any)

Ham22, 2017-03-06:
Is this to test an upgrade mechanism or just to reliably re-flash the devices? If the later then just use the programmer?

Ham22, 2017-03-15:
The script gives 100% success if it all goes well. The issue is if something goes wrong (e.g. me ctrl-c) then the programmer becomes unusable until you un/plug the usb cable. This is no good for an automated test system. Some possible suggestions of how we could get around it [here](http://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line).

manohar-narkhede-imgtec, 2017-03-27:
If the ctrl-c is pressed in between the script the issue of JAVA crash was observed sometimes. The immediate flash fails but after that  the clicker gets flashed. Otherwise the script seems to be working which is confirmed with Ian.

closed 2017-03-27 by manohar-narkhede-imgtec