CreatorDev/contiki #77: Add uncrustify checks in JenkinsFile

abhijit-mahajani-imgtec, 2017-03-03:
As mentioned in https://github.com/contiki-os/contiki/blob/master/CONTRIBUTING.md, we can add uncrustify checks either in git pre-commit hooks but still it would be better to have the check in Jenkinsfile too so that it gets caught in CI.

Shpinkso, 2017-03-03:
Need some acceptance criteria in this ticket

mtusnio, 2017-03-09:
Check here - https://github.com/CreatorDev/contiki/tree/3.0-mikro-e/tools/code-style - for all the tools to fix/check for issues. I think the acceptance criteria could be as follows:

- Fix all issues with uncrustify in the files we have changed in cpu/pic32 & platform/mikro-e
- Add uncrustify checks to the Jenkinsfile for any changed files in a PR